aronstrandberg/mvk15 #14: Convert SQL table schema into create table query.

karlek, 2015-02-17:
Convert this table schema into a PostgresSQL create table query. 

[image]

fellan12, 2015-02-17:
I can take this assignment :D

fellan12, 2015-02-17:
I have created the tables from the scheme. And when creating the tables i used these type for the attributes. Have i missed something or should i change something?

```SQL
CREATE TABLE DroneModel (
ID INTEGER PRIMARY KEY NOT NULL UNIQUE, 
Model TEXT, 
MaxWeight FLOAT);

CREATE TABLE Drone (
ID INTEGER PRIMARY KEY NOT NULL UNIQUE, 
ModelID INTEGER REFERENCES DroneModel (ID));

CREATE TABLE DroneData (
ID INTEGER PRIMARY KEY NOT NULL UNIQUE, 
DroneID INTEGER REFERENCES Drone (ID), 
LocationID INTEGER UNIQUE,
Timestamp TIMESTAMP);

CREATE TABLE Location (
ID INTEGER PRIMARY KEY REFERENCES DroneData (LocationID),
Longitude FLOAT,
Latitude FLOAT);
```

karlek, 2015-02-17:
Is there a geo-location data type in PostgresSQL?

fellan12, 2015-02-17:
Timestamp is the closest one i can find, but there are a lot or types.
Check here if there is something else that works better: http://www.postgresql.org/docs/9.4/static/datatype.html

aronstrandberg, 2015-02-17:
Points?
http://www.postgresql.org/docs/9.4/static/datatype-geometric.html#AEN6730

fellan12, 2015-02-18:
Changes SQL tables to following:
```SQL
CREATE TABLE DroneModel (
ID INTEGER PRIMARY KEY NOT NULL UNIQUE, 
Model TEXT, 
MaxWeight FLOAT);

CREATE TABLE Drone (
ID INTEGER PRIMARY KEY NOT NULL UNIQUE, 
ModelID INTEGER REFERENCES DroneModel (ID));

CREATE TABLE DroneData (
ID INTEGER PRIMARY KEY NOT NULL UNIQUE, 
DroneID INTEGER REFERENCES Drone (ID), 
LocationID INTEGER UNIQUE,
Timestamp TIMESTAMP);

CREATE TABLE Location (
ID INTEGER PRIMARY KEY REFERENCES DroneData (LocationID),
gps POINT);
```

fellan12, 2015-02-18:
Change SQL code to naming convensions:

```SQL
CREATE TABLE drone_model (
id INTEGER PRIMARY KEY NOT NULL UNIQUE, 
model TEXT, 
maxweight FLOAT);

CREATE TABLE drone (
id INTEGER PRIMARY KEY NOT NULL UNIQUE, 
model_id INTEGER REFERENCES drone_model (id));

CREATE TABLE drone_data (
id INTEGER PRIMARY KEY NOT NULL UNIQUE, 
drone_id INTEGER REFERENCES drone (id), 
location_id INTEGER UNIQUE,
timestamp TIMESTAMP);

CREATE TABLE location (
id INTEGER PRIMARY KEY REFERENCES drone_data (location_id),
gps POINT);
```

fellan12, 2015-02-18:
Made further changes to the SQL:
```SQL
CREATE TABLE drone_model (
id INTEGER PRIMARY KEY, 
model VARCHAR(30), 
max_weight FLOAT);

CREATE TABLE drone (
id INTEGER PRIMARY KEY, 
model_id INTEGER REFERENCES drone_model (id));

CREATE TABLE drone_data (
id INTEGER PRIMARY KEY, 
drone_id INTEGER REFERENCES drone (id), 
location_id INTEGER NOT NULL UNIQUE,
timestamp TIMESTAMP);

CREATE TABLE location (
id INTEGER PRIMARY KEY REFERENCES drone_data (location_id),
gps POINT);
```

fellan12, 2015-02-18:
Removed underscore because it doesnt show up on postgres for some reason....

```SQL
CREATE TABLE dronemodel (
id INTEGER PRIMARY KEY, 
model VARCHAR(30), 
maxweight FLOAT);

CREATE TABLE drone (
id INTEGER PRIMARY KEY, 
modelid INTEGER REFERENCES dronemodel (id));

CREATE TABLE drone_data (
id INTEGER PRIMARY KEY, 
droneid INTEGER REFERENCES drone (id), 
locationid INTEGER NOT NULL UNIQUE,
timestamp TIMESTAMP);

CREATE TABLE location (
id INTEGER PRIMARY KEY REFERENCES dronedata (locationid),
gps POINT);

```

fellan12, 2015-02-18:
Last changes with working underscores:

```SQL
CREATE TABLE drone_Model (
id INTEGER PRIMARY KEY, 
model VARCHAR(30), 
max_weight FLOAT);

CREATE TABLE drone (
id INTEGER PRIMARY KEY, 
model_id INTEGER REFERENCES drone_Model (id));

CREATE TABLE drone_Data (
id INTEGER PRIMARY KEY, 
drone_id INTEGER REFERENCES drone (id), 
location_id INTEGER NOT NULL UNIQUE,
timestamp TIMESTAMP);

CREATE TABLE location (
id INTEGER PRIMARY KEY REFERENCES drone_Data (location_id),
gps POINT);

```

fellan12, 2015-02-19:
Is this issue done or is there more stuff to do?

karlek, 2015-02-19:
It's done, looks great! Closing this issue.

closed 2015-02-19 by karlek