aronstrandberg/mvk15 #27: Construct a list of programmers in this project

karlek, 2015-02-26:
It's easier for me to assigne issues to people if I know who wants to code.

Write in this issue if:
* you want to code,
* how much you want to code,
* how much of the tutorials you've read or
* if you got any preferences.

Sorry for that badly structured list :camel:

aronstrandberg, 2015-03-04:
- I want to code
- I don't know. "Lagom"
- Been going through both D3 and Rails, not finished

sandrahbackstrom, 2015-03-05:
Jag är på!

sandrahbackstrom, 2015-03-05:
I want to code
I want like little/a few assignments because of SRS and reports
Will be doing some Rails tutorials

fellan12, 2015-03-05:
I want to code
I want "Lagomt" with assignments
Been doing tutorials on fundamentals on js, ruby and python
Want to work on the pie 2, less/nothing on rails

nejlika, 2015-03-11:
Since we have received the hardware a list of dedicated programmers would present a more solvable situation for delegation.

nejlika, 2015-03-17:
I could help out with coding but will have a rather busy 4th period so my time will be limited. I've done parts/all of the Getting Started with Rails.

agnesaman93, 2015-03-19:
I want to code.
Little - medium
Focus on frontend
Have not done any tutorials yet, but were planning to look at js this week.

mengelin, 2015-03-19:
Pretty much exactly like Agnes. Leaning towards frontend, but anything is fine.

RasmusFredrikson, 2015-03-19:
I want to code.
Little - Medium
Focus on Raspberry Pi
Haven't dont any tutorials yet, planning on doing them next week

mengelin, 2015-03-19:
Nevermind what I might have said earlier. I want to code the shit out of that Raspberry Pi

nejlika, 2015-03-27:
@shappe @Patstrom Någon input?

RasmusFredrikson, 2015-04-09:
The programming assignments has been assigned to the project members.

closed 2015-04-09 by RasmusFredrikson