aronstrandberg/mvk15 #28: Investigate D3.js

karlek, 2015-02-26:
Is it applicable to our frontend?

Ref:

http://d3js.org/

karlek, 2015-03-04:
Progress?

nejlika, 2015-03-05:
Current issue that Aron doesn't have access to a substatial source of GeoJSON data, could be solved once we have access to the GPS.

nejlika, 2015-03-11:
We have recieved input from HiQ regarding the use of D3.js, they seem to be adamant that we use D3.

karlek, 2015-03-18:
To further specify what HiQ wants to use D3.js for: each node in our map view should be clickable and information about the current, previous and next nodes should be used to compute some values and be displayed with D3.js

nejlika, 2015-03-27:
To add on the issue, check wether or not we can expose CartoDB to D3.
http://cartodb.com/

sandrahbackstrom, 2015-05-04:
Can close?

closed 2015-05-04 by sandrahbackstrom