aronstrandberg/mvk15 #31: Formula to calculate velocity and acceleration

aronstrandberg, 2015-03-06:
We need a formula to calculate velocity and/or acceleration (and anything else interesting we can, maybe x- and y-axis velocity?) based on only current **x- and y-axis gps coordinate** and **timestamp**. This is ready to implement in the prototype, so let's get it done so we can have it ready in time for the meeting with HiQ!

Assigned @mengelin but if anyone else feels up to it, go!

fellan12, 2015-03-06:
You cant calculate velocity, yet acceleration based on only one set of GPS points, you will need atleast two sets for velocity and 4 sets for acceleration.

aronstrandberg, 2015-03-06:
Sorry, current and all previous is what I mean of course

karlek, 2015-03-18:
Closely related to #36

sandrahbackstrom, 2015-05-04:
Irrelevant as we use the speed function in the GPS? Can close?

closed 2015-05-04 by sandrahbackstrom