aronstrandberg/mvk15 #33: Only pan camera when drone reaches edge

aronstrandberg, 2015-03-08:
Right now the camera pans to the new dot every time one is drawn, making for slow performance and hacky animations. Only panning when really necessary might improve this.

karlek, 2015-03-09:
By camera, are you referring to the viewport of the Google maps iframe?

Is there a possibility to check the where in the viewport our coordinate tag will created? If so, then we can check the distance from the border of the viewport to the tag and if it's bigger than a certain value we make a smooth transition to the new position.

karlek, 2015-03-09:
Haha is referring to SO a bad habit? Anyways found this _untested_ code snippet for getting the long / lats of the viewport. 

https://stackoverflow.com/questions/14285963/google-maps-get-viewport-latitude-and-longitude

aronstrandberg, 2015-03-09:
Yes, exactly. I'm pretty sure this is trivial and that the API has built-in methods to do this, I just wrote it up so I don't forget it.

karlek, 2015-05-06:
@aronstrandberg implement today? I'm in META at 12 today.

aronstrandberg, 2015-05-06:
It's difficult to test this without new data points streaming over time, so we either need that to work or bring back the faked ones

closed 2016-03-19 by aronstrandberg