aronstrandberg/mvk15 #36: Multiple point visualisation

nejlika, 2015-03-11:
Magnus requested functionality for visualization of correlation and connections between adjacent datapoints. 

One suggestion is to use a group of 3 points and have a separate popup-window display their inherited qualities - such as velocity and acceleration.

karlek, 2015-05-04:
@aronstrandberg, we have to remember this.

karlek, 2015-05-04:
When a large point is hovered it's given color / and / the previous small dots and the last big dot / and / the next small dots and the next large dot is given color.

karlek, 2015-05-06:
@aronstrandberg should we make this a version 2 issue? Since our current solution we'll need significant modding to implement this.

aronstrandberg, 2015-05-06:
Do we? Shouldn't it be pretty straight forward, just modify the style-feature function to also style the next & previous points?