aronstrandberg/mvk15 #38: Create UML-diagrams

nejlika, 2015-03-18:
The SRS requires a visual representation of the system. Björn recommended, and I agree, using UML.

Create a draft of controlflow for both front-end and back-end which can be improved iteratively.

This is a issue directed at @aronstrandberg and @karlek

karlek, 2015-03-18:
Was the UML-lecture any good?

aronstrandberg, 2015-03-19:
Was anyone there?

fellan12, 2015-03-23:
Aron have u made your sequence diagram yet?

RasmusFredrikson, 2015-03-27:
All the UML-diagrams are done. Closing the issue.

nejlika, 2015-05-06:
Update UML:s to reflect the current structure:
@karlek: Backend, [old](https://files.slack.com/files-pri/T03HNB08S-F0443N7BB/backend_-_page_1.png)
@fellan12: Raspberry Pi, [old](https://files.slack.com/files-pri/T03HNB08S-F04474ML1/mvk_raspberry_pi_uml.png)
@aronstrandberg: Frontend, [old](https://files.slack.com/files-pri/T03HNB08S-F044N0JMR/untitled_document_-_new_page.png)

karlek, 2015-05-06:
Hai! Does anyone remember what site we used to make these? Looks cleaner with consistent design

nejlika, 2015-05-06:
@karlek Lucidcharts maybe?

karlek, 2015-05-06:
Ty.

RasmusFredrikson, 2015-05-10:
The UML-diagrams are done. Closing.

closed 2015-05-10 by RasmusFredrikson