aronstrandberg/mvk15 #41: Create a script to database

fellan12, 2015-03-23:
Create a python script that make a insertion query to the database, without using ssh.

aronstrandberg, 2015-03-23:
Why not SSH?

fellan12, 2015-03-23:
I think it's better if we can insert directly to the database. Having SSH requires us to hard code the VM's login information and password, with in my head could cause some security issues.

fellan12, 2015-03-23:
Martin, consult with aron on how to proceed involving the script.

fellan12, 2015-03-23:
@mengelin did you get this!

nejlika, 2015-03-24:
Aren't all the security regarding login solved by the whitelist?

mengelin, 2015-03-24:
Umm no Felix. It doesn't require us to hard code the login information into the script. You can just enter it into the terminal as a part of the script. I hardcoded it into the script simply to save time when testing it.

mengelin, 2015-03-24:
We could use VNC to control the pi while flying. Starting and stoping a lap aswell as using ssh to the VM. Right?

nejlika, 2015-03-27:
Yes @mengelin, that could be a solution. How stable is VNC in the wild?

mengelin, 2015-03-30:
Wouldn't be any different from what it is when we have tested it in Meta, since we are using wi-fi hotspot for VNC

sandrahbackstrom, 2015-05-04:
done?

closed 2015-05-04 by sandrahbackstrom

karlek, 2015-05-04:
Done!