aronstrandberg/mvk15 #42: Figure out a way to attach the all equipment to the drone

fellan12, 2015-03-23:
We need to figure out a way to attach all our gadgets to the drone and still make it fly.

With our previous testing on the matter we found out these problem that have to be solved.
* Balance issues, when the placement of the load is incorrect. (Not center of gravity)
* Weight, with only the Pi and the battery the drone could barely take flight.

karlek, 2015-03-24:
If we remove the protective case and change to a smaller battery would that solve the problem?

fellan12, 2015-03-24:
I guess is we remove the case it would help. But if we change the battery to a smaller one, we need one that has enough power to start the pi.

karlek, 2015-03-24:
The battery only needs 1 hour of power (although that would be quite on the small end), but where do we find a battery like that?

karlek, 2015-03-24:
Is this the current setup?
Case: 47g (http://www.adafruit.com/products/2258)
Raspberry PI B+: 42g (http://www.adafruit.com/products/1914)
GPS: 8.5g (http://www.adafruit.com/product/746)
Battery pack: 140g (without cables) (http://www.adafruit.com/products/1565)
Wifi USB: 6g (http://www.adafruit.com/products/1012)

fellan12, 2015-03-24:
Yes and apparently it's almost to much...

fellan12, 2015-03-24:
or just about too much...

karlek, 2015-03-24:
We can, in an emergency, try to remove a battery cell from the battery to make it lighter and smaller. If it doesn't lift, we don't need the battery anyway

karlek, 2015-03-24:
Is it too much even without the case?

fellan12, 2015-03-24:
haven't tried that yet.

closed 2015-03-27 by nejlika

karlek, 2015-03-30:
Just clarifying for sake of future references: we split the battery into a single battery cell to reduce the weight