aronstrandberg/mvk15 #50: Write PHP program

sandrahbackstrom, 2015-04-22:
Convert data from database to geoJson

sandrahbackstrom, 2015-04-22:
Shappe and I are working on this. 

How should we convert the data to geojson? manually? 
How should the connection be between PHP and website? 

We need the correct database properties to be able to connect, set up the correct database.

karlek, 2015-04-22:
You could search for a PHP library which already converts to GeoJSON. 

When you ask 'how should the connection be', are you asking how to connect to the database with PHP?
If so there are multiple ways, the best is PDO.
https://php.net/manual/en/book.pdo.php

What do you mean about correcting the database properties?

sandrahbackstrom, 2015-04-22:
There is no geoJson convertion function. There is however a Json function. I figure that is not what we want. If you look at the code, its on github, we have tried to do it manually.

The connection, as in should we push or should JS ask for data. 

We have updated the database, adding one more attribute (read) from the database in SRS1. Look at the code and you'll see why and how we use it.

aronstrandberg, 2015-04-22:
I think the best way is JS will poll for new data using ajax. @karlek?

karlek, 2015-04-22:
I google'd and there seems to be a few unoffical PHP libraries for GeoJSON conversion however since my knowledge about GeoJSON is lacking I'll need to read about about them first.

The best way, as you said @sandrahbackstrom and @aronstrandberg is for the JS to ask/pull the data.
(In PHP you could use $_GET requests for this)

Ah, I see. We probably want a reset feature for the read-attribute when we are testing, should a make an issue for that? Or is it easier to do it manually from the shell?

karlek, 2015-04-22:
Well your solutions seems to be working and looks clean!

sandrahbackstrom, 2015-05-04:
Done!

closed 2015-05-04 by sandrahbackstrom