aronstrandberg/mvk15 #61: Remove bars as the number increases?

aronstrandberg, 2015-05-04:
Looks very crowded when the number of bars becomes high. Will possibly improve when the height of the bars isn't random

karlek, 2015-05-04:
Possible solutions, only the latest n-points. Only every n-th point of all points. Random n points. etc.

karlek, 2015-05-04:
It's quite bad if the bars move / change during flight if you want to hover over them... I can imagine it would be quite frustrating.

karlek, 2015-05-04:
Fixed number of bars. In the beginning a few bars will have info and the rest will be a very low value and a different color. Then they will rise up bullshit etc.

The selection of the points should be last 20. Future update might change this selection or offer a setting.

nejlika, 2015-05-05:
If we choose a solution where we only plot the latest n points, there should be an option to display the whole lap after it's completion.

aronstrandberg, 2015-05-05:
y u gon b such a bitch

nejlika, 2015-05-05:
'cuz if I don tell u, hiq be mad

aronstrandberg, 2015-05-05:
[image]

karlek, 2015-05-05:
@nejlika we mean in the graph view, not the plotted points.

karlek, 2015-05-06:
Not prio atm.

sandrahbackstrom, 2015-05-08:
Could be scrolling the graph from left to right and right to left instead of removing

closed 2016-03-19 by aronstrandberg